NordicPlayground/nRF51-ble-bcast-mesh #108: Vagrant Build environment

Protoneer, 2016-05-17:
Does anyone have a Vagrant File available for building this project with GCC?

Protoneer, 2016-05-17:
I tried a setup from Adafruit (https://github.com/adafruit/ARM-toolchain-vagrant/) and that might be a good start.

I take it I need to install the SDK but I am not really sure wher to put it.

Protoneer, 2016-05-17:
Okay... So we managed to get it working. 

The confusing bit is having the SDK in a folder below this project folder. (In our case we had the SDK folder in our root Vagrant folder and manually moved the code from this repository to the Examples folder)

To make this integrate better with GIT we have two choices that I can think of. Create a new project just for the nRF GCC build environment(cloning this repository into its example folder ) or change the make files with a fixed but changeable SDK location and have more freedom with where projects can sit.

The last option feels cleaner... Any opinions or input?

trond-snekvik, 2016-05-18:
Hi there, 

I would agree on the second option - make an SDK path variable and be done with it. The chosen structure for the example projects is really just to make Keil easy to use and setup. It can't do macros or environment variables in the paths, so we had to make sure everyone cloned the mesh with the SDK in a known relative path. The GCC makefiles included in the projects have [an `SDK_BASE` variable](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/examples/BLE_Gateway/gcc/makefile#L19) pointing to the SDK, so they're technically using your second option already :)