NordicPlayground/nRF51-ble-bcast-mesh #109: radio tx power

czb, 2016-05-19:
I study the code and hit the following line

        NRF_RADIO->TXPOWER      = ((RADIO_TXPOWER_TXPOWER_0dBm << RADIO_TXPOWER_TXPOWER_Pos) & RADIO_TXPOWER_TXPOWER_Msk);

in `radio_control.c`. Does that mean that the mesh uses 0dbm? If so, can it be changed (to e.g. +4dbm)? If, not, why not? :)

trond-snekvik, 2016-05-19:
This is correct :) You can change it to +4dbm if you want, just by changing this value. The downsides are increased power consumption (although not really significant when comparing it to the RX power), and increased packet collision ratio for devices that wouldn't be interfering with each other on 0dm. The upside is of course longer range :)

czb, 2016-05-20:
Would you mind making the parameter changeable so it is not necessary to touch the inner code? :wink:

trond-snekvik, 2016-05-20:
Sure :)

trond-snekvik, 2016-06-30:
Hi @czb, can you verify that the addition to the interface in 0.8.4 works for you? :)

trond-snekvik, 2016-12-01:
Verified on our side for 0.8.7. Please reopen if there are still issues related to this.

closed 2016-12-01 by trond-snekvik