NordicPlayground/nRF51-ble-bcast-mesh #111: Problem with trickle

victorpasse, 2016-05-25:
Hi, Sometimes trickle->i becomes much larger than the curren time, I have been able to find that in order_next_transmission in the "if (timeout < ts_begin_time + time_now + 1000)"
timeout > ts_begin_time is sometimes true, this makes 
vh_order_update(timeout - ts_begin_time);
become called with negative time but the argument is unsigned so it overflows.
What can I do to fix this?

trond-snekvik, 2016-05-26:
Oh, yes. The update-function can handle inaccuracies in the timestamp, so changing 

    if (timeout < ts_begin_time + time_now + 1000)
    {
        vh_order_update(timeout - ts_begin_time);
    }

to

    if (timeout < ts_begin_time)
    {
        vh_order_update(0);
    }
    else if (timeout < ts_begin_time + time_now + 1000)
    {
        vh_order_update(timeout - ts_begin_time);
    }

should do the trick.

victorpasse, 2016-05-26:
Good, this was the fix i tested, it seems to work good.

victorpasse, 2016-05-26:
Just to verify, the whole if state ment should look like this:
	uint64_t ts_begin_time = timeslot_get_global_time();
	if(timeout < ts_begin_time)
	{
		vh_order_update(0);
	}
	else if (timeout < ts_begin_time + time_now + 1000)
	{
		vh_order_update(timeout - ts_begin_time);
	}
	else
	{
		timer_order_cb(TIMER_INDEX_VH, timeout - ts_begin_time, transmit_all_instances);
	}

trond-snekvik, 2016-05-26:
Yep, looks good :+1:

trond-snekvik, 2016-06-30:
As the underlying timer changed for 0.8.4, this should no longer be a problem. All timing is now independent of timeslots. Feel free to reopen if this turns out to cause any further issues.

closed 2016-06-30 by trond-snekvik