NordicPlayground/nRF51-ble-bcast-mesh #112: fifo with size!=2^n

victorpasse, 2016-05-25:
Hi, I want a fifo size of 40 for several of the fifos but do not want to waste ram to have 64.
What do i need to change for this to work?
Is it only where you do operations like this:
& (p_fifo->array_len - 1)
or are other parts of the code dependent of size=2^n?

trond-snekvik, 2016-05-25:
Hi,
This is possible, but requires some changes to the fifo implementation. To understand why, we have to understand how index overflow works in the fifo:

Instead of rolling over the head and tail indexes when they hit fifo length, the fifo implementation assumes that indexes will naturally overflow on a multiple of the fifo size. "Dumb" fifo implementations keeps the indexes inside the fifo size. This forces us to waste a slot in the fifo, as the case where `head == tail` will be ambiguous (is the queue empty or full?), and we have to say that the fifo is full when `tail == head - 1`. If we let the indexes wrap at `fifo_size * N` (for `N>1`), and modulo the "raw" index on the fifo size on lookup, we can get rid of the ambiguity, as the fifo is guaranteed to be empty when `head == tail`, and full when `head == tail + fifo_size` (see the defines at L39 and L40 in [fifo.c](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/fifo.c#L39)).

There are two problems with doing non-power of two:

1. You have to do modulo instead of bitwise and to get the buffer index, which will be significantly slower. 

2. You have to solve index rollover, as it will no longer be aligned with the natural rollover of the data type.

An alternative solution would be to bite the bullet, and keep the indexes inside the fifo size, then do `if (index > size) index -= size;` on each iteration. This costs you a slot in the fifo, but if you're hitting 40+ slots, it doesn't really matter that much.

trond-snekvik, 2016-12-01:
Closing due to inactivity

closed 2016-12-01 by trond-snekvik