NordicPlayground/nRF51-ble-bcast-mesh #118: Updates have broken branch bl_app

pjyoung1963, 2016-06-15:
It seems that the update to rev 1748 changed the event structure rbc_mesh_event_type_t
This breaks compilation of several files in the bl_app branch
mesh_aci.c, mesh_gatt.c, rbc_mesh.c, transport_control.c, version_handler.c
since these are still using the older structure.

trond-snekvik, 2016-06-15:
Yes, this has not been addressed yet. A lot of the examples and code not related to the bootloader is still broken in this branch. I'll push some updates on this branch sometime this week. [The readme in the bl_app branch](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/tree/bl_app#about-this-branch) has a warning about the state of the branch along with some additional information about its purpose. Sorry for the inconvenience.

pjyoung1963, 2016-06-15:
Thanks,

The fixes were fairly simple anyway.

I’m more interested in this as an example for how to handle the packet Tx / Rx for the mesh since the trickle algorithm will not meet my requirements, I also do not need to expose my data via GATT, but the rest of the code appears really useful.

 

I see that there are 4 separate aspects of the project which it would be nice to treat independently although I have not yet got into the detail of how to separate them.

 

1.       Mesh transport, i.e the sending and receiving of frames using advertising packets, the soft device does not really provide integrated support for this and your timeslot implementation is a very useful example.

2.       Mesh packet handling, i.e  deciding the packet contents, handling the packets to maintain a database and deciding what to send and when, here I have very different requirements and both the database format and the trickle algorithm are inappropriate

3.       Mesh database interface, you provide GATT and Serial interfaces which I need to disable

4.       DFU support, which is completely independent of the mesh data and forwarding algorithms.

 

But the example you provided looks like it will be very helpful anyway, so thank you.

 

Regards

 

Phil.

trond-snekvik, 2016-06-29:
Hi Phil, sorry about the delay here.

We have been looking into splitting up these modules, as we definitely agree that they could be useful on their own. While the serial and DFU can be disabled by removing their defines (`RBC_MESH_SERIAL` and `MESH_DFU`), and GATT can be disabled by disabling [this directive](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/mesh_gatt.c#L33) (alterations to the framework:( ), the transport level of things is not as available or generic as it could have been. 

We have been doing some experiments with the lower layers in other projects (that will become available in a few months), that we hope to pull into the OpenMesh. We see the usefulness in that module as a standalone component, and it's nice to get confirmation that it's something that's worth pursuing :+1: