NordicPlayground/nRF51-ble-bcast-mesh #121: Alternative Access Address > Motivation?

olsky, 2016-06-28:
May I ask the motivation behind the alternative access address? 

It can be very useful, but I'd like to know if I get it the right direction...

thank you!

trond-snekvik, 2016-06-29:
Hi,

The alternative access addresses allows us to scan for regular BLE advertisement packets, and use the new :star2:[peek functionality](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/rbc_mesh.h#L591):star2: to discover beacons or other non-mesh devices while the mesh runs on an alternate access address. In addition, the DFU runs exclusively on the standard access address, to allow interoperability between adjacent networks/subnets.

olsky, 2016-06-29:
packets peek: long expected feature! :) :+1:

closed 2016-12-01 by trond-snekvik