NordicPlayground/nRF51-ble-bcast-mesh #123: DFU Documentation

olsky, 2016-06-28:
Could you please add the detailed DFU documentation? :-)

In the 0.8.4 I've found code with channel_map, is this move towards multi-channel DFU / Mesh?

Thank you

trond-snekvik, 2016-06-29:
I'm not able to do this this week, but I'll try to make it as soon as possible, it's been pending for some time :)

The channel map is to cater to multi channel DFU, yes. The mesh bitfield is locked to single channel right now, as the mesh devices aren't capable of jumping between channels while receiving. Our experiments don't actually show any reason to do multichannel meshes at this point, as the potential performance increase coming from dodging particularly noisy channels is eclipsed by the overhead of jumping. Also, multiple meshes in the same area performs significantly better when running on different channels.

olsky, 2016-06-29:
Thank you Trond!

I was just curious to know if the multi-channel was the way to go.

Looking forward to details of dfu. Thank you!