NordicPlayground/nRF51-ble-bcast-mesh #130: TX_Power not consistent

olsky, 2016-07-13:
The new set/init with tx_power interface has no effect.

Current flow:

- radio_init is called from transport_control.c TX_power is set to 0dBm
- later setup_event is called, there TX_power is set with event's field

Somehow it does not affect the power.

Could it be that an extra param needed in radio_init function with the tx_power? Thus the power will be constant?

trond-snekvik, 2016-08-16:
Hi Olsky, looks like this one slipped through the cracks, sorry about that.

I'll take a look at it for the next release. I have a couple of additional fixes coming in September.

olsky, 2016-08-16:
Hi Trond, thank you for note!

Looking forward for new commits :+1:

downwith, 2016-08-19:
Just stumbled across this issue on my own, too.  Can confirm.  The event didn't seem to be having an effect for me at all.  Just the radio_init hardcoded value.

trond-snekvik, 2016-12-01:
Fixed in 0.8.7.

closed 2016-12-01 by trond-snekvik