NordicPlayground/nRF51-ble-bcast-mesh #131: How to get started with BLE Mesh ?

anujdutt9, 2016-08-15:
Hi. I have the NRF51 Dongles and the nrf software as well. I want to create a simple Mesh. How can I go about it ? Where to start from ??

trond-snekvik, 2016-08-16:
Hi,

The simplest way to get started is to download the [nRF Connect Mesh application](https://devzone.nordicsemi.com/blogs/998/nrf-connect-mesh-beta-release-making-ble-mesh-deve/). It allows you to easily flash and communicate with the open mesh, and will help you learn the concepts.

Once you feel familiar with the mesh, and want to start developing your own applications, take a look at the [BLE gateway example](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/tree/master/nRF51/examples/BLE_Gateway). It displays some basic functionality, and should give you some insight to how to use the APIs. Also make sure you read the readme, as it contains some vital information on setting up the repo.

If you want to explore the internals of the mesh, there are a couple of additional guides in _/docs/_:

- [How it works](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/docs/how_it_works.adoc) - concepts and internal mechanisms

- [Usage](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/docs/usage.adoc) - detailed interface descriptions

Happy meshing!

anujdutt9, 2016-08-17:
Hey, thanks for the information. I'll get started with it : ) :+1:

closed 2016-10-05 by daviddedwin