NordicPlayground/nRF51-ble-bcast-mesh #133: Do All nodes know about all other nodes ?

anujdutt9, 2016-08-17:
Hi. I have 4 nrf51 dongle and was playing around with them using the nordic mesh. One thing that I noticed is that once the mesh is formed, each node only has information of the other node to which it is directly connected. Is there a way to let each node know about what all nodes are present in the mesh ? Is there any command you have for that ?
Thanks : )

chriswillecke, 2016-08-17:
From what I have seen there isn't a command built in to the firmware. Part of the reason for this the mesh is connectionless, and relies on blindly broadcast advertisements for communication. There is no confirmation of reception by any other nodes when a command is sent. This makes it difficult to know when a node has joined or left the mesh. A last-seen or timeout type scheme could be implemented. For example in the ping pong example, m_data[0]-m_data[6] are populated with the local Bluetooth address which is random so should be unique. Using this information you can determine which nodes are actively broadcasting on the mesh and when the value was last updated. This should give you enough information to determine presence of a node in the mesh.

mrquincle, 2017-05-22:
@anujdutt9 See https://github.com/crownstone/bluenet/blob/master/docs/PROTOCOL.md#mesh-message for a possible implementation of a protocol on top of Open Mesh. We've opted for a `Source address` to indeed be able to obtain this type of information.

closed 2024-08-19 by anujdutt9