NordicPlayground/nRF51-ble-bcast-mesh #138: Library compatibility with Keil μV5?

bayou9, 2016-10-13:
To deploy an example project (e.g. ping pong), is a manual re-configuration (modifying include paths, mainly) required?         Or this step is completely unecessary, and any problem encountered when attempting to re-build a project regarding include paths is a result of Kei lμV5 not properly installed and setup?

If it's the former, please shed some light on what packs/library were you using when developing this project, because there are seemingly some minor issues with its compatibility with nRF_Drivers 3.1.0 (mostly nrf_drv_uart, missing a few MACRO definitions). 

Please advise, thank you.

trond-snekvik, 2016-10-20:
Hi,

If your Keil project has trouble finding SDK files, you might have the project cloned in the wrong directory, or you might be linking the wrong version of the SDK. Can you verify that your setup matches the instructions in the [Getting started section](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh#getting-started) of the readme?

bayou9, 2016-10-21:
Hello, you were right, I've placed the sdk files at the wrong directory. It turned out I have to place it 2 levels above the ble-mesh-xxx directory. All things were resolved, thank you.

closed 2016-12-01 by trond-snekvik