NordicPlayground/nRF51-ble-bcast-mesh #143: nRF51 S130 support?

thedjnK, 2016-11-19:
I previously tried to get this working for the nRF51 using the S130 softdevice (I haven't actually been able to get it working [the BLE gateway] for nRF51 S110 or nRF52 S132) and have noticed that 10 days ago the S130 nRF51 text was removed from the readme - does this mean it didn't support/support for S130 has been removed?

trond-snekvik, 2016-11-29:
Hi there, could you describe the issues you have with the BLE Gateway? Is this related to #145?

The S130 is still supported, this looks like a mistaken documentation change on our part.

thedjnK, 2016-11-29:
The two tickets are unrelated. With the precompiled flash files if I flash them then nothing happens. IfI build the application with DFU support enabled and go through the steps to flash it then nothing happens (on nRF52), the nrfutil program gives a single 'failed to create process' error message which looks to be because it's a python stump and none of the python code is included. I was unable to get the mesh branch of nrfutil to compile on windows so tried the master branch but the commands have all changed so it can't be used. I got the mesh branch compiling and running on Linux but it was not able to communicate with the module (these are Laird BL600/BL652 modules not the Nordic nRF51/52 development kits) so there is no 4th button to press.

Additionally I tried flashing the application from nrfgo studio but it would never run either, I'm convinced this is related to the bootloader because I removed the MESH DFU build flag and flashed just the soft device and application and it works fine, I did try playing around a bit loading the DFU firmware like this without the bootloader and eventually got it to boot with the error checking disabled whereby it failed with the error message the the command handler couldn't be found which makes sense. I gave up with the nRF51 and have moved forward with the nRF52 now.

trond-snekvik, 2016-11-30:
I see. Looking at the [datasheet](http://cdn.lairdtech.com/home/brandworld/files/User%20Guide%20-%20BL600-DVK%20(Board%20Revision%205).pdf) for the module you're using, it looks like it's using a different set of pins for UART than what the bootloader is compiled for. These settings come from the SDK board-files, which are included in this project [here](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/tree/master/nRF51/SDK/bsp), and are fed into the hardware [here](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/serial_handler_uart.c#L224-L227). You can either change them inline, or create a custom board-file (perhaps your vendor provides one?). In any case, the bootloader must be recompiled.

Are you able to run the nrfutil script if you clone the branch and install it with pip from your local copy? This is the most reliable way to use it on my side, but there have been several reports related to py2exe creating trouble for the installation.

Some additional notes:
- nrfGo Studio is unable to flash the bootloader correctly, and is not recommended. (It fails to flash the `NRF_UICR->BOOTLOADERADDR`)
- If the bootloader and device page is flashed correctly, you should be able to run the application with bootloader, regardless of the serial pin issue.

thedjnK, 2016-11-30:
The BL600 (nRF51) module uses a different pinout but I was also trying on the BL652 (nRF52) which has the same UART pinout as the nRF52 development board (from what I remember reading). I cloned the branch and built it, the filesize is about 10MB but there's a message printed showing there's a lot of missing libraries, I'll have a look on my development machine sometime and try again.

I tried both nrfgo studio and the flash batch file, both gave the same result. I also tried running it directly through keil with a breakpoint on the first line of code but it would never get there, I haven't tried seeing if running it in debug with keil works without DFU, I'll also try that.

trond-snekvik, 2016-12-01:
I looked into the modules you're using some more, and I found at least [one version](http://www.mouser.com/ds/2/223/Datasheet%20-%20BL652-963794.pdf) of the kit that has different pins. I'm not sure if this matches yours, but it's worth double-checking. For reference, the PCA10028/PCA10031/PCA10040 kits run this config:

Pin | Number
----|-----
RTS | 8
TXD | 9
CTS | 10
RXD | 11

Also, here's the requirements list for nrfutil:
- Python 2.7
- pyserial >= 2.7
- enum34 >= 1.0.4
- click >= 6.0
- ecdsa >= 0.13
- behave >= 1.2.5

I'll do some investigation of the nrfutil and nrfjprog scripts in this repository, they're creating more confusion than they're providing solutions at this point.

For the `NRF_UICR->BOOTLOADERADDR` problem, you can verify that this has register has been properly flashed by calling nrfjprog: `nrfjprog --memrd 0x10001014`. The result should be `0x3A000`. A typical symptom of the nRFGo Studio problem is that this register remains unflashed, in which case this readout returns `0xFFFFFFFF`.