NordicPlayground/nRF51-ble-bcast-mesh #144: documents for Bandwidth_test example

rick1082, 2016-11-28:
Hi,

Is there any documents or guides for nRF51-ble-bcast-mesh/nRF51/examples/Bandwidth_test ?
For now, the description of the Bandwidth_test example still remains in BLE_Gateway,
It's a little hard to figure out how to use this example to perform bandwidth test.

Thanks :)

baruaanomadarshishuvro, 2016-11-29:
Hi,
read me doc has been updated for the bandwidth test .

Thanks :)

rick1082, 2016-11-29:
Thank you @baruaanomadarshishuvro,  According to the document, there still needs a python test script. Would Nordic provide the example of the python test script ? Really thanks :D

baruaanomadarshishuvro, 2016-11-29:
Hi, 
i have added a sample python script(Bandwidth_Test.py) to calculate the bandwidth and also updated the read me doc accordingly.

Thanks :)

rick1082, 2016-11-29:
So many thanks, @baruaanomadarshishuvro 
Really appreciate : D

closed 2016-11-29 by rick1082