NordicPlayground/nRF51-ble-bcast-mesh #160: ~70ms mesh interrupt

nachonov, 2017-01-25:
Hi,

I got this quoted from issue #137 :

> The mesh also does some work in the highest IRQ priority, but this is limited to simple state changes, and shouldn't block you for more than 50us at a time.

I am trying to generate a PWM signal on 24 different GPIOS. All PWM signal work on the same frequency 200Hz and 50 different divisions for setting the duty cycle. So that implementation will create a timer interrupt every 5ms. My understanding from the previous quote is that the mesh will hold the micro for a max of 50us which is acceptable for my application. However, I have found that there are some long interrupts of around 70ms

[image]

I am not sure if the mesh is causing this, but I have not implemented nothing else but the timer interrupts to handle the PWM signals.

Any help most welcome

trond-snekvik, 2017-01-26:
Hey, this is interesting. I've not been able to reproduce on my side yet, so I have some questions about your application.

- Are you on nRF51 or nRF52?
- Which IRQ level are you running your PWM controller in?
- Do you run any Softdevice connections or advertisements?
- Which timer do you use to wake up your processor?
- How often do you see this? Is it related to some specific interaction with your mesh network?

Some suggestions:
- Run your PWM code in the APP_HIGH IRQ level (1 on nRF51, 2 or 3 on nRF52).
- Don't use NRF_TIMER0 or NRF_RTC0, as the Softdevice controls them.

nachonov, 2017-01-26:
Hey trond. Thanks a lot for your interest and help. I will try to answer your question:

> Are you on nRF51 or nRF52?

nRF51. Specifically I am using the nRF51 DK PCA10028 v1.1.0

> Which IRQ level are you running your PWM controller in?

NVIC_SetPriority(TIMER1_IRQn, NRF_APP_PRIORITY_HIGH);

> Do you run any Softdevice connections or advertisements?

I am connected to the Master Control Panel in order to send some commands. I have set the connection interval to 100ms on the MCP.

> Which timer do you use to wake up your processor?

NRF_TIMER1

> How often do you see this? Is it related to some specific interaction with your mesh network?

It does appear at random times. The length is always the same. Around 70ms. I am not sure about that. 

I have include the module library and the main that I have implemented. There are no many comments as under development but it is a very simple code so far.

[PWM_LEDs.zip](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/files/732525/PWM_LEDs.zip)