NordicPlayground/nRF51-ble-bcast-mesh #174: not working example with nRF51422

MateuszKSS, 2017-03-31:
Hello,
I already use SoftDevice S110 v8.0.0 for nRF51 and SDK v.8.1.0 for nRF51 and i want to run example BLE Gateway but it cant work.
In Keil we have seen window "Programming Done" but BLE Scanner cant see it.
Where's the problem?

trond-snekvik, 2017-03-31:
Hi,

I'm afraid I need some more info to help you.

Which target are you building? serial or not? with button support? 
Which board are you using?
Have you tried connecting a debugger? Is the program running?
Have you tried scanning for it with the [nRF Connect app for Android](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) or something similar for iPhone?

Try compiling and flashing the "nRF51422 xxAC buttons" target. If you have a Development Kit, does it change its LEDs if you press any of its buttons? Are there any LEDs lit when you start up?

MateuszKSS, 2017-03-31:
Hi,

Thanks for repply.

I just solved this problem.

 

Best regards / Pozdrawiam / Grüße / Un Saludo
Mateusz Miga 

Electronic Designer / Konstruktor Elektronik
Tel: +48 32 32 46 397 

 Stopka <file:///c:\podpis\stopka.jpg> 

  _____  

with nRF51422 (#174)

 

<https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=
en>  Connect app for Android or something similar for iPhone?

<https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/issues/174#issu
ecomment-290696764>  it on GitHub, or mute
T-kfOlaks5rrO7fgaJpZM4MvbaM>  the thread.
<https://github.com/notifications/beacon/AZkmlQ6tk-wbH6eMPYscK6s7ci0BOarCks5
rrO7fgaJpZM4MvbaM.gif>

closed 2017-03-31 by trond-snekvik