NordicPlayground/nRF51-ble-bcast-mesh #175: nRF52 not receiving mesh events

mjdietzx, 2017-03-31:
Hey!

I'm building an application based on the BLE Gateway example and never receiving any mesh events (never reach this code: https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/examples/BLE_Gateway/main.c#L309).

Don't have a great setup so haven't been able to test very thoroughly and haven't narrowed down the issue: i just know that my application never receives a mesh event (not even after initializing the mesh). Any known issues here? And have these examples been tested well for nRF52?

I'm guessing this call to `nrf_adv_conn_init();` was added because `RBC_MESH_EVENT_TYPE_INITIALIZED` is not being received?
- https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/examples/BLE_Gateway/main.c#L264
- https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/examples/BLE_Gateway/main.c#L167

Building with gcc, mesh dfu not enabled, and in node mode (not serial) in case any of that info helps.

Thanks,
Mike

thedjnK, 2017-04-02:
Are you using a softdevice and is it the right version? If you're not using a softdevice you can't use the BLE example. The `nrf_adv_conn_init()` code is calling softdevice functions to start a GATT server and advertise so that central devices can connect to the service. Yes it works fine on nRF52 devices (I have it working on Laird BL652 development kits and Nordic nRF52 development kits without issue) using GCC and keil. I was never able to get DFU working however.

If you want to just test if it's working and assuming you have two Nordic nRF52 development kits you can flash that to both of them and pressing one of the buttons should change the LED status on the other board.

If you can, flash your code and use gdb to see what's happening, use the bt (backtrace) command to see where the code is and if it's got stuck (if it enters an error handler then it will sit there forever).

mjdietzx, 2017-04-02:
Using sdk v12.0.0 and s132 v3.0.0. BLE gateway is working (device advertises and i can connect and discover services upon startup) and mesh actually seems to be working. the problem is i'm not seeing any rbc mesh events being received (i.e. rbc_mesh_init() succeeds but i never see RBC_MESH_EVENT_TYPE_INITIALIZED)

thedjnK, 2017-04-02:
You only get a callback with that status if you have the serial interface enabled and don't have a bootloader, see `rbc_mesh/src/mesh_aci.c` under `case SERIAL_CMD_OPCODE_INIT:`. You won't see any events if you're just using one device and updating the values on that device via BLE, you need another node which will receive the events. For example if you have 2 nodes, you connect to one with a phone and change a handle's value, the value will update on that node (you can change the code that runs when this happens, it's in `rbc_mesh/src/mesh_gatt.c` in function `mesh_gatt_sd_ble_event_handle` just under `case MESH_GATT_EVT_OPCODE_DATA:`) and will be transmitted out to the other nodes, on the other nodes you will get an event with `RBC_MESH_EVENT_TYPE_CONFLICTING_VAL`, `RBC_MESH_EVENT_TYPE_NEW_VAL` or `RBC_MESH_EVENT_TYPE_UPDATE_VAL` depending on if it already has the handle used or not. You can get a callback for `RBC_MESH_EVENT_TYPE_TX` on the node your phone is connected to if you enable transmission events for the particular handle using `rbc_mesh_tx_event_set(handle, true);`, you will get a callback any time that handle is transmitted (i.e. you will get it a lot)