NordicPlayground/nRF51-ble-bcast-mesh #180: OTA for mesh nodes

edwardcchan, 2017-04-20:
I would like to know the procedures for OTA update for the mesh node. Shall I flash all the blank nodes with the procedures you stated in the DFU update as the basis? Thanks

trond-snekvik, 2017-04-25:
Yes. All devices needs a Softdevice, a bootloader, a device page, and optionally an application. Have you tried following the steps in the [Quick start guide](../blob/master/nRF51/bootloader/dfu_quick_start.adoc)?

edwardcchan, 2017-06-07:
I have tried to follow the Quick Start Guide. I have faced the following problems: 

--- 

1. When I tried to use to run the nrfutil.exe within the pc-util folder, I got error message

"failed to create process."

I have no clue on what it means. 

--- 

2. So I used the nrfutil.exe installed by python pip to try Step 10: 

.\pc-util\nrfutil dfu genpkg --application ../examples/BLE_Gateway/bin/rbc_gateway_example_serial_blinky_PCA10028_nRF51422_xxAC.hex --company-id 0x00000059 --application-id 1 --application-version 2 --key-file private.txt --sd-req 0x0064 --mesh dfu_test.zip

However, nrfutil complained that some arguments are not available, for example, 

Error: no such option: --mesh

--- 

Thanks for your attention.