NordicPlayground/nRF51-ble-bcast-mesh #182: Notification not enabled for sdk13 and nRF52832

codezork, 2017-04-28:
Hi Trond,
after some search why the code doesn't hvx to a host device, I found that in mesh_gatt.c in function mesh_value_char_add the characteristic write type must be with response in order to get notifications enabled by the host. So line 294 for  branch development must be:
`ble_char_md.char_props.write = 1;` instead of 
`ble_char_md.char_props.write_wo_resp = 1;`. 
Can you confirm my finding?
Cheers,
Joerg

trond-snekvik, 2017-06-29:
From what I remember, this shouldn't be related, since the write_wo_resp is for host to mesh communication, while hvx is enabled by setting notify to 1 on the line below. 

Do you get any change in hvx behavior from this?

codezork, 2017-07-01:
Hi,
 
Looks like that SDK13 needs write and notify to be true to work properly and responding to notification. With this change it worked for me, without service discovery of the host didn't recognize the notification link. 
Great work by the way!
 
Best regards
Joerg Riegel / codezork
  _____