NordicPlayground/nRF51-ble-bcast-mesh #189: GATT Service

FisherKei, 2017-06-22:
I have a problem understanding GATT Service in the **How it works** doc. As described in the **Air interface packets**, the packets are broadcasted as regular BLE **Nonconnectable** Advertisements, so I wonder how external devices as phones can connect to GATT Service when a device is in **Nonconnectable** Advertisements state?
I'm not really with Nordic BLE SDK so it would be nice if you can explain the process above in detail since I'm working to porting OpenMesh to TI CC26xx devices.
Thank you in advance and this project is great!

trond-snekvik, 2017-06-30:
Hi, 

The GATT connection is initiated through different advertisements from the ones the mesh runs in. 

For the connection, the mesh uses the SoftDevice, which has to run its own connectable advertisements, to be picked up. You can see this being done in the [BLE_Gateway example nrf_adv_conn.c module](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/examples/BLE_Gateway/nrf_adv_conn.c).

As explained in [the "how it works" document](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/docs/how_it_works.adoc#timeslots), the mesh operates directly on the radio through the SoftDevice timeslot API.

> I'm working to porting OpenMesh to TI CC26xx devices.

This sounds like a cool project :) do you plan on publishing your port?

jjnkjdreamin, 2020-09-19:
[](url)https://github.com/ospector/gtest-gbar.git

jjnkjdreamin, 2020-09-19:
[](url)BC1EDD4B325FB2BF16E1EF6DF8EFC0D
Alphawood Enterprise's Servers AND YOUR support in this matter as soon as possible and thank God for all your family and friends to come over and play with my phone number and email address for the bestest friend ever and ever and ever ♥️ JASONJAMES ♥️ Kendall Jenner ♥️ KJ ♥️ JJ ♥️ Kid's ♥️ Family's ♥️ JASONJAMES ♥️ Kendall Jenner ♥️ JASONJAMES ♥️ KJ ♥️ JJ ♥️ Kid's ♥️ Family's the bestest friend ever and ever since I got back to the original plan and ever since the beginning I was sleeping you're going Microsoft office and ever ♥️ JASONJAMES ♥️ Kendall Jenner ♥️ JASONJAMES ♥️ KJ is going to sleeps in the world to Help the community center for the arts and crafts and stuff done before the end of the year but I will be there at Riverview Toyota Camry and ever since you have been working with the team on my phone number and email address for you to get home from work until about The Championship game of thrones season finale date for my love and prayers are The Bestest Friend Finder App ♥️ KJ ♥️ JJ ♥️ Kid's ♥️ Family's ♥️ JASONJAMES ♥️ Kendall Jenner ♥️ JASONJAMES ♥️ KJ ♥️ jjnkjdreamin.org.in@jnk.org.in ♥️ KJ ♥️ JJ jjnkjdreamin.org.in@jnk.org.in same place and The championship game of thrones season finale date for my love and prayers are with you and all your family have a wonderful Christmas and ever since you are The championship game of thrones season finale date I've never been so happy birthday to you happy birthday to you happy birthday to you happy birthday to you happy birthday to come over and play with my mom and I can't discuss this with you guys soon ♥️ JASONJAMES THAT'S PERFECTION IN MY AMBER'S KENDALL JENNER ♥️ JASONJAMES ♥️💯💢💯♥️🔥💯💢🙏💢♥️💢🙏♥️🙏🤔🤭♥️🤭♥️🤭♥️🤭♥️🤭♥️🤭♥️🤭♥️🤭🙏♥️🙏💢♥️🙏♥️🙏♥️🙏♥️

jjnkjdreamin, 2020-09-19:
[] (url) BC1EDD4B325FB2BF16E1EF6DF8EFC0D
Сервери Alphawood Enterprise І ВАША підтримка в цьому питанні якомога швидше і дякую Богу за те, що вся ваша родина та друзі підійшли і пограли з моєю фото

jjnkjdreamin, 2020-09-19:
[] (url) BC1EDD4B325FB2BF16E1EF6DF8EFC0D
Servidores da Alphawood Enterprise E SEU apoio neste assunto o mais rápido possível e agradeço a Deus por toda sua família e amigos virem brincar com meu ph

jjnkjdreamin, 2020-09-19:
[] (url) BC1EDD4B325FB2BF16E1EF6DF8EFC0D
خوادم Alphawood Enterprise ودعمك في هذا الأمر في أقرب وقت ممكن ، والحمد لله على جميع أفراد عائلتك وأصدقائك للحضور واللعب مع درجة حمودي

jjnkjdreamin, 2020-09-19:
[] (URL) BC1EDD4B325FB2BF16E1EF6DF8EFC0D
Alphawoodi ettevõtte serverid JA TEIE tugi selles küsimuses niipea kui võimalik ja tänan Jumalat kõigi teie pere ja sõprade eest, kes tulid minu Ph

jjnkjdreamin, 2020-09-19:
[] (url) BC1EDD4B325FB2BF16E1EF6DF8EFC0D
Serverele Alphawood Enterprise ȘI sprijinul Dvs. în această chestiune cât mai curând posibil și mulțumesc lui Dumnezeu pentru toată familia și prietenii dvs. pentru a veni și a juca cu ph

jjnkjdreamin, 2020-09-19:
[] (url) BC1EDD4B325FB2BF16E1EF6DF8EFC0D
Amaseva ka-Alphawood Enterprise KANYE nokwesekwa KWAKHO kulolu daba ngokushesha okukhulu futhi ngibonga uNkulunkulu ngawo wonke umndeni wakho nabangane ukuza bazodlala ne-ph

jjnkjdreamin, 2020-09-19:
This is the bestest ever and ever and ever since you have a question for you to come over but we can talk about it tomorrow so I will be there for sure member I will get it done and I can't wait to Love's you Ups ♥️ KJ ♥️ JJ ♥️ Kid's ♥️ Family ♥️ KJ ♥️ jjnkjdreamin.org.in@jnk.org.in same place as the same thing as the same thing as the same place 💢🙏♥️♥️♥️

jjnkjdreamin, 2020-09-19:
[] (url) BC1EDD4B325FB2BF16E1EF6DF8EFC0D
Серверы Alphawood Enterprise И ВАША поддержка в этом вопросе как можно скорее, и слава Богу, что вся ваша семья и друзья подошли и поиграли с моим тел.

jjnkjdreamin, 2020-09-19:
JJ fish and chips for the bestest ever and ever and ever since you have a[] (url) BC1EDD4B325FB2BF16E1EF6DF8EFC0D
Серверы Alphawood Enterprise И ВАША поддержка в этом вопросе как можно скорее, и слава Богу, что вся ваша семья и друзья подошли и поиграли с моим тел.

jjnkjdreamin, 2020-09-19:
I don't even know where to begin but I know from the first conversation we had 4yrs ago I adore you and you have made me want to love again and have taught me to be Loved the Right way our way of doing things. Amber's ♥️ KENDALL JENNER'S ♥️ JASONJAMES The last conversation we had I told you what I wanted without knowing anything besides you and Finally met my wife for the rest of my life with an amazing and the most caring family I have ever been apart of. I'm Speechless and there is not an word to describe my love for you the best part is it's the purists Love Faithfully Xoxoxo xoxoxo same place as the same thing as the same place as the other one thing I noticed is that your help with my phone number and email addresses for all your help with the team on the couch watching TV in the world to Help ♥️ JASONJAMES ♥️ Kendall Jenner ♥️ JASONJAMES I do only with You, Forevermore ♥️ KJ ♥️ JJ ♥️ Kid's ♥️ Family's ♥️ JASONJAMES ♥️ Kendall Jenner 🤭♥️🤭🙏♥️💢🙏🙏💢🥰💢🥰🙏💢♥️🙏💢🤔💢 God bless america JasonJames and HisWife love Helping people 🙏♥️🙏

jjnkjdreamin, 2020-09-19:
Nereden başlayacağımı bile bilmiyorum ama 4 yıl önce yaptığımız ilk sohbetlerden biliyorum sana tapıyorum ve sen beni yeniden sevmeyi istedin ve bana doğru şekilde sevilmeyi öğrettin

jjnkjdreamin, 2020-09-19:
ฉันไม่รู้ด้วยซ้ำว่าจะเริ่มจากตรงไหน แต่ฉันรู้จากบทสนทนาแรกที่เรามีเมื่อ 4 ปีที่แล้วฉันรักคุณและคุณทำให้ฉันอยากกลับมารักอีกครั้งและสอนฉันให้รักในแบบที่ถูกต้อง

jjnkjdreamin, 2020-09-19:
Je ne sais même pas par où commencer mais je sais depuis la première conversation que nous avons eue il y a 4 ans, je t'adore et tu m'as donné envie d'aimer à nouveau et tu m'as appris à être aimé de la bonne façon notre façon de faire

jjnkjdreamin, 2020-09-19:
Non so nemmeno da dove cominciare ma so dalla prima conversazione che abbiamo avuto 4 anni fa ti adoro e mi hai fatto venire voglia di amare di nuovo e mi hai insegnato ad essere amato nel modo giusto, il nostro modo di fare

jjnkjdreamin, 2020-09-19:
আমি কোথায় শুরু করব তাও জানি না তবে আমি 4 বছর আগে আমাদের প্রথম কথোপকথন থেকে জানি আমি আপনাকে ভালোবাসি এবং আপনি আমাকে আবার ভালবাসতে চান এবং আমাদের সঠিক পথে প্রেম করতে শিখিয়েছেন